shebuildsrobots/remy_model #1: Awesome Project!

bengalaviz, 2024-10-12:
Can you give more detail on the project? Wiring? Battery? Final build pictures? 

I would like to make one for my son for school. Thanks!

BigPappy098, 2024-10-14:
This is AMAZING! I too would like updated info on parts used besides the servos. Would also LOVE the code for movement with the accelerometer. Keep up the amazing work!!

shebuildsrobots, 2024-10-15:
I added the accelerometer code and a few photos. Hope this helps! Leaving the issue open in case I have time in the future to add more wiring and battery details.

jversloot, 2024-10-16:
Would you be able to add a quick photo of the circuit and which microcontroller you used? Thanks!

shebuildsrobots, 2024-10-16:
@jversloot I'm on mobile but hopefully this will help until I can sit down at a computer:

Controller: Adafruit Micro Mini
Pinouts: 5 and 7 to servo data pins (any PWM pin is fine)
GND -> GND and VCC -> 5V for each servo 

If you are using the MPU6050 accelerometer version:
SDA on accelerometer to A4
SCL on accelerometer to to A5
VCC on accelerometer to 5V
GND on accelerometer to GND

Mimia2709, 2024-10-22:
@shebuildsrobots are you able to provide links for these supplies? Thank you!

AdiWA14, 2024-10-25:
@shebuildsrobots If you were to use 2 accelerometers for both hands, would you connect the other accelerometer to other analog pins (A2,A3)

PaulElrich, 2024-10-26:
@AdiWA14 

They go into the same pins, the difference comes from where the power is connected to the accelerometer board.

https://m.youtube.com/watch?v=t25uTeo3cJo