shebuildsrobots/remy_model #4: Stringy

Mimia2709, 2024-10-18:
Hi! I saw your video on Instagram and I love it so much. I’ve attempted to print Remy multiple times (photos attached) but I’m having a really hard time getting it to work. I have a Bambu Lab P1S 3D Printer. I’ve seen that once it gets to the nose area, the filament becomes stringy and won’t form as supposed to. Is there anyway you can help me with this? My last try was the one with the tree supports and I’m not sure that worked much better. Thanks. 

[image]
[image]
[image]

shebuildsrobots, 2024-10-18:
Hi! I haven't heard reports of this happening for anyone else -- maybe someone else browsing will have advice? I sliced my model with UltiMaker with Supports toggled on, and printed using the Engineering setting. I hope that helps troubleshoot. Good luck!

brandanop, 2024-10-18:
I printed mine successfully on a Bambu labs A1 with AMS for multicolor. I would recommend making sure your filament is dry and running the calibration. I also used the same supports. If that doesn’t work, maybe try rotating it to make Remy lay on his back with supports?
[image]

bengalaviz, 2024-10-20:
I have a Prusa MINI and had to print with supports also. 
[image]

Mimia2709, 2024-10-22:
Thank you so much. I got it to work!

closed 2024-10-26 by Mimia2709