shebuildsrobots/remy_model #7: Need help

Mimia2709, 2024-10-27:
I bought a 170 tie points mini breadboards by inland and an adafruit metro mini 328 V2. I also purchased a couple other wires. I also have the servos you linked. I need more help with the coding and wiring. @shebuildsrobots i really want this to be my Halloween costume so please help me!!!

Mimia2709, 2024-10-27:
[image]

shebuildsrobots, 2024-10-27:
Hi! What specific questions do you have? If this is your first time working with servo motors, check out https://shebuildsrobots.com/project_tea.html for an explanation.

Mimia2709, 2024-10-27:
@shebuildsrobots  So I think my question is how do I wire the servos to the mini breadboard? I have a photo attached of what I got together so far 

[image]

Also, what kind of program do I put the code you provided in?

shebuildsrobots, 2024-10-28:
I can't be sure from the picture, but it looks like your board still needs to be soldered to the header pins that came with the board so that you have a good electrical connection. You will probably get wonky results until you solder them together.

You can upload the code using the Arduino IDE, which is free to download from arduino.cc.

For your servos, wire as following:
Red wire -> plug this into the same row on the breadboard as the 5V pin on the Metro
Brown wire -> plug this into the same row on the breadboard as the GND pin on the Metro
Yellow/orangeish wire -> plug this into the same row on the breadboard as either pin 5 or 7 on the Metro

Hope this helps!